vortigont/pzem-edl #1: Значение потребленной мощности в киловатт часах, сбросилось и начало считать заново

Aleks130699, 2022-01-30:
{"age":245,"U":200.1,"I":10.79,"P":2104,"W":16976,"hz":50.0,"pF":0.97}
Вчера значение когда подключил esp значение показало 52квч, хотя на прошивке тасмота было 378 квч, сегодня же значение пошло считать заново

Aleks130699, 2022-01-30:
Да скорее всего это проблема переполнения переменной, посчитал что 378 квт, это как раз 5 полных раз по 65,5 квт, и остается примерно 50 квт

vortigont, 2022-01-30:
ОК, понял, похоже проблема вот здесь.
```
        energy  = __builtin_bswap16(*(uint16_t*)&value[REG_ENERGY_L*2])  | __builtin_bswap16(*(uint16_t*)&value[REG_ENERGY_H*2]   << 16);
```
 Сначала нужно делать байт своп, а потом сдвиг, поправлю тогда вечерком.

vortigont, 2022-01-30:
P.S. хороший у вас расход :)

Aleks130699, 2022-01-30:
16 бит для этих чисел скорее всего не хватает, как по мне, ну вам лучше знать)

vortigont, 2022-01-30:
само значение энергии 32 битное, но пзем отдает его как два 16 битных регистра с перевернутым порядком старшего байта. Я ошибочно сдвиг под скобку внёс ) поправил - тестируйте.

closed 2022-01-30 by vortigont

Aleks130699, 2022-01-31:
Проверил, все круто, теперь правильно показывает

commit 04ff361d (2022-01-30): fix byteswap-after-shift call for High 16 bits of the 4 byte metrics

Closes #1