vortigont/pzem-edl #2: Missing UART library?

MBL-derek, 2022-02-02:
Trying to get your code to compile.

```
msgq.hpp:14:10: fatal error: driver/uart.h: No such file or directory
```
https://github.com/vortigont/pzem-edl/blob/9750b664b68908340d9679b8b6ee936e260e693c/src/msgq.hpp#L14

What am I missing library-wise?

vortigont, 2022-02-02:
Hi! ugh, that's strange. uart.h is the header from esp32 IDF sdk, no need for any additional libs.
What is your build env? Plain IDF/Arduino/platformio?
I have examples for the lib based on Platformio, it build fine with a simple 'pio run'.
Could you check it?

vortigont, 2022-02-02:
The only external dependency is my fork of a [LinkedList](https://github.com/vortigont/LinkedList) library, needed for TimeSeries structures

closed 2022-02-09 by vortigont