vortigont/pzem-edl #3: esp32/spiram.h is missing

Alaa-Aim, 2022-04-06:
@vortigont , after updating to new version of this library, I get an error while compiling "In file included from espem/espem.h:12:0,
                 from espem/interface.cpp:1:
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:32:26: fatal error: esp32/spiram.h: No such file or directory
compilation terminated.
In file included from espem/espem.h:12:0,
                 from espem/espem.cpp:9:
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:32:26: fatal error: esp32/spiram.h: No such file or directory
compilation terminated.
*** [.pio\build\espem\src\interface.cpp.o] Error 1
*** [.pio\build\espem\src\espem.cpp.o] Error 1
In file included from espem/espem.h:12:0,
                 from espem/main.cpp:11:
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:32:26: fatal error: esp32/spiram.h: No such file or directory
compilation terminated.
*** [.pio\build\espem\src\main.cpp.o] Error 1
============================================================ [FAILED] Took 8.03 seconds ============================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:08.027
====================================================== 1 failed, 0 succeeded in 00:00:08.027 ====================================================== "

and an error in the main.cpp also as shown in the screen shot below 

[image]

vortigont, 2022-04-06:
ah, yes, this header has been changed in a recent IDF 4.x and platformio still uses an older IDF. I will add some include guard for this.
Also better to move to a recent core with platformio, I'll provide an example config for espem.

Alaa-Aim, 2022-04-06:
@vortigont many thanks for your efforts

commit 7f38aff7 (2022-04-06): fix spiram include for older IDF core (<4.0.0)

Closes #3
closed 2022-04-06 by Alaa-Aim