vortigont/pzem-edl #18: esp32/spiram.h: No such file or directory

rodolpheAV, 2024-07-16:
Hello, 

I'm trying to build your project with arduino IDE as I'd like to use it on a ESP32 C6 which is not yet supported by platformio.
When building I'm getting the following error : 

```
In file included from C:\Users\Rodolphe\Documents\Arduino\libraries\pzem-edl\src\timeseries.cpp:13:
..\pzem-edl\src\timeseries.hpp:33:10: fatal error: esp32/spiram.h: No such file or directory
   33 | #include <esp32/spiram.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
```
It seems to me that esp32/spiram.h may have moved (or been removed) in the [last version of the framework](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s2/migration-guides/release-5.x/system.html#psram)

vortigont, 2024-07-17:
Hi! Thanks for highlighting! Could you try with a recent commit if it works properly, pls?

rodolpheAV, 2024-07-17:
Hi and thanks for the quick update. 
Copying the last commit files in my src folder, I do not see this error anymore, but I'm getting some new ones which seems quite strange as these are syntax errors : 

`In file included from C:\ESP-32\WattMeter\resources\pzem-edl\examples\pzem_cli\src\timeseries.cpp:13:
timeseries.hpp:281:20: error: expected unqualified-id before ')' token
  281 |     TSContainer<T>(){};
      |                    ^
In file included from C:\ESP-32\WattMeter\resources\pzem-edl\examples\pzem_cli\src\LList.h:22,
                 from C:\ESP-32\WattMeter\resources\pzem-edl\examples\pzem_cli\src\pzem_edl.hpp:18,
                 from C:\ESP-32\WattMeter\resources\pzem-edl\examples\pzem_cli\src\pzem_edl.cpp:13:
LinkedList.h:23:32: error: expected ')' before '*' token
   23 |         ListNode<T>(ListNode<T>* _next = nullptr) : next(_next) {}
      |                    ~           ^
      |                                )
LinkedList.h:24:21: error: expected unqualified-id before 'const'
   24 |         ListNode<T>(const T& _v, ListNode<T>* _next = nullptr) : data(_v), next(_next) {}
      |                     ^~~~~
LinkedList.h:24:21: error: expected ')' before 'const'
   24 |         ListNode<T>(const T& _v, ListNode<T>* _next = nullptr) : data(_v), next(_next) {}
      |                    ~^~~~~
      |                     )
exit status 1
`
I might do something wrong.
Any thoughts about it ?

vortigont, 2024-07-17:
Ugh... I should have got rid of that linked-list dependency long ago...
Let me rework it in a while.

vortigont, 2024-07-17:
@rodolpheAV 
Could you, pls try [arduino3](https://github.com/vortigont/pzem-edl/tree/arduino3) branch?
I do not have access to pzem hardware now, so could only do build test.
If that works fine with real device, then I will merger this change.
Tnx!

Jason2866, 2024-07-17:
@vortigont Excuse for hijack this issue. Are you online on Discord? -> Platformio...

vortigont, 2024-07-17:
@Jason2866 sorry, time to sleep in my time zone. Will join tomorrow.

rodolpheAV, 2024-07-17:
@vortigont I'm now able to build and flash the ESP.
I've been testing with the pzem_cli to see if I can get any feedback from the PZEM. I can see the CLI menu, send commands but I cannot get any message from the pzem. The ESP32 Is actually acting as if no pzem was connected at all. 

That said, I may be connecting it wrong as I'm new with this hardware.

I've been trying : 
PZEM 5v <--->ESP 5v
PZEM Rx <--->ESP Tx
PZEM Tx <--->ESP Rx
PZEM Gnd <--->ESP Gnd

D2 led of the PZem is on, but I cannot boot the ESP32 with these wiring, even reseting it do not output anything on the monitor.

I've been trying : 
PZEM 5v <--->ESP 3.3v
PZEM Rx <--->ESP Tx
PZEM Tx <--->ESP Rx
PZEM Gnd <--->ESP Gnd
No led is on on the Pzem (except the D3 led during the reset buton press), I can boot the menu on the ESP but cannot get any answer interacting from the menu.

I've been trying with another ESP32-Wrover and I'm getting the same issue.

Hope this can help

vortigont, 2024-07-18:
@rodolpheAV you might need [hw mod](https://github.com/vortigont/pzem-edl/blob/main/docs/hwmod.md) to make it work with 3.3 volts. 5 volts usually works fine, but be careful to pick gpio's, some of those might be sensible to input voltage and affect esp to boot. You can pick those that are mentioned in examples.
Also make sure that PZEMS's AC main is connected - it does not work without being supplied with AC power.

rodolpheAV, 2024-07-22:
@vortigont after some pin switching, I can confirm that I'm now able to communicate with the Pzem and get calbacks and coherent data from it. I'll dig a bit more in your lib to ajust it to my needs but it seems to me that your new branch is properly working with a real device.

Thanks for your help

vortigont, 2024-07-22:
thanks for testing!
Happy coding with pzem-edl :)

PR #19 (2024-07-22): Arduino3
 - support building on Arduino 3.x core
 - remove 3rd party LinkedList lib dependency

Closes #18